parallaxinc/PropWare #21: Make sometimes fails on Windows systems

DavidZemon, 2015-02-01:
Due to this error case
http://hdrlab.org.nz/articles/windows-development/make-interrupt-exception-caught-code-0xc00000fd-addr-0x4217b/

DavidZemon, 2016-01-30:
I've not found a reliable work around for this issue. Users will be on their own to find a Make executable that works for their system. Another option is to use PropWare within SimpleIDE, which is now supported via the download instructions.

closed 2016-01-30 by DavidZemon

bweir, 2016-01-31:
What version of make is this a problem with? I've never encountered this before and I run 64-bit windows. I'd recommend that your users use mingw-w64 or msvc (which includes nmake) for the build environment, because I have observed that the sort-of default mingw32 that everyone uses isn't really maintained anymore, which could be a contributing factor to this.

DavidZemon, 2016-01-31:
I hit the problem originally with the original [GNU Make for Windows](http://gnuwin32.sourceforge.net/packages/make.htm), which hasn't been updated since 2006.

I then tried Cygwin and MinGW (don't recall if 32- or 64-bit) and hit problems there as well. Basically, I hate Windows and decided it was easier to call this "unsupported" and provide SimpleIDE instructions. If a user gets it to work on Windows, I'm thrilled, and I encourage them to use it. However, I'm no longer interested in chasing down this bug.