parallaxinc/PropWare #22: Add Sudo Install Option

trezitorul, 2015-02-12:
When running the INSTALL.py script on linux it would be nice if one could have the option of installing cmake and propgcc into files belonging to root without having to run the whole install script as root (which causes problems later). For example the script will throw an error if you attempt to install propgcc into /opt/parallax currently. It would be nice if the script would ask for your sudo password to execute the install command for propgcc and cmake.

DavidZemon, 2015-02-21:
Good idea. I think I know how I can implement this too. Thanks for the suggestion

DavidZemon, 2015-08-02:
A better solution has been implemented which does not require the Python Script! yay! :)

The downside is, PropGCC is no longer installed

closed 2015-08-02 by DavidZemon