parallaxinc/PropWare #33: Create a JSON library

DavidZemon, 2015-03-30:
As sparked by [this forum thread](http://forums.parallax.com/showthread.php/160630-C-Is-it-possible-to-send-a-struct-serially-without-requiring-parsing-on-the-receiver?p=1323511&viewfull=1#post1323511), create a library capable of importing/exporting JSON. Might make use of the Printer/scanner class too.

[StackOverflow on JSON/C](http://stackoverflow.com/questions/6673936/parsing-json-using-c)

[jsmn](http://zserge.com/jsmn.html) is an existing, small JSON library. Maybe a custom module isn't necessary, or maybe jsmn could be the start and then tweaked to reuse some of PropWare's existing code.

bweir, 2016-01-31:
Whaa? They want this for Propeller?

DavidZemon, 2016-01-31:
Ha! No one has explicitly asked for it. I just thought it would be really cool to have it. Maybe someone will use it some day? Maybe not? Henceforth why it's on the "someday" milestone rather than a specific release.

commit db484641 (2017-07-16): [Closes #33] Include ArduinoJson library in installation package
closed 2017-07-16 by DavidZemon