parallaxinc/PropWare #52: Watchdog Object

DavidZemon, 2015-11-19:
A common peripheral on other microcontrollers is the "watchdog" which will reset the CPU if the watchdog is not fed periodically. A watchdog object in PropWare would be a great addition.

pmrobert, 2015-11-19:
That would be a very small object unless I'm misinterpreting your message.
I usually just code it in one of the free-running loops; it's only like 5
lines of code? BTW, thank you for PropWare!

    -Mike

DavidZemon, 2015-11-19:
It might well end up being a very small object - I'm okay with that. Care to post code for your idea?

And glad you like it! I haven't any questions on it in a very long time so I was wondering if anyone was still using it.

DavidZemon, 2015-11-20:
Here's what I'm imagining:

* Watchdog class inherits from the [Runnable](http://david.zemon.name/PropWare/classPropWare_1_1Runnable.html) interface
~~~~~~~~~cpp
namespace PropWare {
  class WatchDog : public Runnable {
    public:
      WatchDog(const unsigned int timeout, /* params for Runnable */)
        : Runnable(/* params for Runnable */),
          m_timeout(timeout) { }

      void reset () {
        this.m_timer = CNT;
      }

      void run () {
        this->m_timer = CNT;

      }

    private:
      const unsigned int m_timeout;
      volatile unsigned int m_timer;
  }
}
~~~~~~~~~

* One of the parameters of the constructor will be a timeout (units probably in clock ticks).
* When the cog starts,

pmrobert, 2015-11-20:
Yes! A way to optionally reset the entire chip would be nice - I usually do
a clkset $7F if the watched cog times out after n tries - the nuclear
option. I know that's not elegant but it is reliable especially if the
watchdog is running in cog only - that's much less likely to get corrupted
by other code writing into hub where they shouldn't be.

DavidZemon, 2015-11-20:
What does `clkset $7f` do?

DavidZemon, 2015-11-20:
`clkset` simply changes the frequency right? That's what I'm seeing from the Spin documentation. But I don't understand what that would do for a watchdog.

Also, looks like I forgot a loop in my code above. Whoops!

DavidZemon, 2015-11-20:
[Thread started](http://forums.parallax.com/discussion/162828/c-c-reset-propeller-via-software) on how to reset the Propeller

pmrobert, 2015-11-20:
Sorry, David, brainfart! it's actual $80,not $7F of CLK register. It does
in fact work as not-so-widely advertised.
 This is the only place I think it's documented:

PropManual v1.2, p28 under CLK register definition for bit 7:

0
Always write ‘0’ here unless you intend to reset the chip.
1
Same as a hardware reset – reboots the chip. The Spin command REBOOT writes
a ‘1’ to the RESET bit.

pmrobert, 2015-11-20:
0
Always write ‘0’ here unless you intend to reset the chip.
1
Same as a hardware reset – reboots the chip. The Spin command REBOOT writes
a ‘1’ to the RESET bit.

DavidZemon, 2015-11-20:
Ha! How cool is that!? :D

pmrobert, 2015-11-20:
Simple nuclear option! :-)

DavidZemon, 2015-11-20:
Okay, so no code as recommended, using the `clkrst` instruction. I'll give this a go soon as I have time. Unfortunately, my time at home has been more sparse than my time at work lately.

~~~~~~~~~~~~~~~~~~~cpp
namespace PropWare {
  class WatchDog : public Runnable {
    public:
      WatchDog(const unsigned int timeout, /* params for Runnable */)
        : Runnable(/* params for Runnable */),
          m_timeout(timeout),
          m_currentCog(cogid()) { }

      void reset () {
        this.m_timer = CNT;
      }

      void run () {
        unsigned int timeSinceLastReset;
        this->m_timer = CNT;

        while (1) {
          waitcnt(SLEEP_TIME + CNT);
          timeSinceLastReset = CNT - this->m_timer;
          if (timeSinceLastReset > this->timeout)
            __builtin_propeller_clkset(0x80); // Hard reset
        }
      }

    private:
      const unsigned int m_timeout;
      const uint8_t m_currentCog;
      volatile unsigned int m_timer;
  }
}
~~~~~~~~~~~~~~~~~~~

pmrobert, 2015-11-20:
That should do it for a full reset. Another variation would be to have the
ability to just stop the dead cog and restart it. As you mentioned earlier
(and you're a whole lot more knowledgeable re the inner workings of PropGCC
than I am) I also wonder if there might be orphaned stack or heap left over
with that scheme and, if so, how it would be freed. I hear you regarding
work/personal/hobby balance, the older I get the more important I realize
that is, I'm glad you realize that too.

    -Mike

DavidZemon, 2015-11-20:
I like the idea of expanding this to have 7 individual watchdogs, or optionally one global.... and it shouldn't require much more space at all to do so.

pmrobert, 2015-11-20:
Yes, you have concisely stated what I wanted to concisely state.

DavidZemon, 2015-11-22:
Feel free to give it a try. The first couple tries were written at a coffee shop without a board, henceforth the re-committing multiple times.

DavidZemon, 2015-12-12:
I've added another issue for expanding to 7 independent timers
https://github.com/DavidZemon/PropWare/issues/56

closed 2015-12-12 by DavidZemon