parallaxinc/PropWare #53: Add video support

DavidZemon, 2015-11-23:
This is huge... but we need VGA support. I don't even have a board that supports VGA though, so this won't happen for a long time unless someone else steps up to the plate.

konimaru, 2015-11-23:
https://github.com/konimaru/waitvid.2048/tree/master/POC/80x25%20RAM/src.side

I know it's just C but it's a start :)

DavidZemon, 2015-11-24:
A few more details can be found in [this thread](http://forums.parallax.com/discussion/162851/object-request-c-vga-support-compensation-negotiable). That does look like a decent start, but I would want those two drivers to be swapped for modifiable code, not binary.

konimaru, 2015-11-27:
FWIW, I only read the forum these days (too much damage to existing posts so why bother).

Anyway, you have sources available but you should accept that **for getting resolution you have to use PASM**. Not entirely sure what the issue with binary only is. I can see that being able to modify the code is useful in the long run but at least you'd have something available now. 32x15 isn't exactly something to write home about :)