parallaxinc/PropWare #60: Instructions for Mac OS X download/installation include cmake install that conflicts with user's current cmake installation

dgately, 2016-01-01:
Provided cmake installation conflicts with user's configuration

On attempting to execute the PropWare download/installation instructions on a recent Mac OS X system (10.11.2 - El Capitan, latest as of 01/01/2016), I've run into a number of issues that I'm working-through in order to assist in your creation of an install package for Mac OS X. Here's an issue about cmake...

Many developer-type Mac OS X users have already installed cmake from Ports, Homebrew or have even done a CMake application installation. These configurations include placing cmake and its utilities into directories that conflict with the PropWare installation provided cmake download. They also cause conflict with PropWare's expected configuration. 

As an example, my current config for cmake places its binaries inside the CMake application bundle (image of the App's directory scheme):

[image]

Other cmake installations place the binaries in /opt/local/bin, /usr/local/bin or /usr/bin, etc.

Its it possible for PropWare to use these alternative configurations, and forgo the cmake.zip install when cmake already exists? I was not easily-able to get PropWare to use my previously installed cmake without replacing several links. On my system I need to preserve the current cmake configuration as many other programs expect the current config.

DavidZemon, 2016-01-01:
First, thank you very much for taking the time to suffer through this. I know its quite rough at the moment.

It sounds like your issue is a question: "Can I place PropWare's CMake installation in another directory?"
If I'm understanding you wrong, please say so.

In short, yes.

Special configuration files are deployed with that bundle though - the magic files that enable using PropGCC with CMake. So any time you invoke CMake for a Propeller-based project, you _must_ use the version of CMake shipped with PropWare. That being said, I did not mess with any existing files in the CMake installation, nor did I even compile a custom version (I just download a binary distro from cmake.org and then add a few files to it). So my instructions say to overwrite your existing CMake install because for 95% of users, it won't do any harm to use the version shipped with PropWare instead of whatever default version there is. PropWare's CMake is still 100% compatible with all of your normal Desktop CMake applications.

If you still prefer not to install into `/usr/local`, simply unzip PropWare's CMake zip into any directory that you like. Once that is complete, you'll need to add the a new environment variable named `PROPWARE_PATH`, which should point to the contents of the main zip file, with the headers and libraries.

The complicated part is ensuring you invoke the right version of CMake for the right projects. I would recommend adding the `bin` directory of PropWare's CMake installation to the front of `PATH`, but that would imply that PropWare's CMake would be invoked for your normal desktop applications too... which I'm assuming _is not_ what you want, or else you would have just followed the normal installation instructions. Perhaps you type out the full path to cmake each time you want to invoke it? Perhaps you can set it permanently in an IDE on a project-by-project basis? Maybe you set an environment variable like `PW_CMAKE` which points to the `bin` directory of PropWare's CMake install, so that you don't have to type out the full path each time??? I'm afraid I don't have a good solution here. If you have ideas, I'm open to them.

With the `PROPWARE_PATH` environment variable set and the correct version of CMake invoked, you should be good to go, with both ZIPs extracted anywhere on your system.

dgately, 2016-01-02:
Ah, I'll work with this! I'll let you know how it goes and report any additional instructions needed. Thanks.

dgately, 2016-01-08:
One thing I notice is that the instructions at http://david.zemon.name/PropWare/Download.xhtml include a download of cmake.zip that does NOT include the actual cmake binaries... Here's what unzip says about the .zip

"$ unzip cmake.zip 
Archive:  cmake.zip
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeCOGCCompiler.cmake.in  
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeCOGCInformation.cmake  
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeCOGCXXCompiler.cmake.in  
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeCOGCXXInformation.cmake  
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeDATCompiler.cmake.in  
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeDATInformation.cmake  
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeDatSymbolConverter.cmake  
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeDetermineCOGCCompiler.cmake  
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeDetermineCOGCXXCompiler.cmake  
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeDetermineDATCompiler.cmake  
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeDetermineECOGCCompiler.cmake  
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeDetermineECOGCXXCompiler.cmake  
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeDetermineSPIN2DATCompiler.cmake  
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeECOGCCompiler.cmake.in  
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeECOGCInformation.cmake  
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeECOGCXXCompiler.cmake.in  
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeECOGCXXInformation.cmake  
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeRunObjcopy.cmake  
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeSPIN2DATCompiler.cmake.in  
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeSPIN2DATInformation.cmake  
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeSpin2DatSymbolConverter.cmake  
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeTestCOGCCompiler.cmake  
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeTestCOGCXXCompiler.cmake  
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeTestDATCompiler.cmake  
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeTestECOGCCompiler.cmake  
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeTestECOGCXXCompiler.cmake  
  inflating: OSXCMake/share/cmake-3.3/Modules/CMakeTestSPIN2DATCompiler.cmake  
  inflating: OSXCMake/share/cmake-3.3/Modules/FindPropWare.cmake  
  inflating: OSXCMake/share/cmake-3.3/Modules/Platform/Generic-gcc-Propeller.cmake  
  inflating: OSXCMake/share/cmake-3.3/Modules/PropellerToolchain.cmake"

There's only the share directory content!

While further instructions also say to execute the following:

   9 sudo ln -s /usr/local/OSXCMake/bin/cmake /usr/local/bin/cmake
   10 sudo ln -s /usr/local/OSXCMake/bin/ccmake /usr/local/bin/ccmake
   11 sudo ln -s /usr/local/OSXCMake/bin/cpack /usr/local/bin/cpack
   12 sudo ln -s /usr/local/OSXCMake/bin/cmake-gui /usr/local/bin/cmake-gui
   13 sudo ln -s /usr/local/OSXCMake/bin/ctest /usr/local/bin/ctest

But there is no "/usr/local/OSXCMake/bin"...

DavidZemon, 2016-01-08:
Thank you for this reminder. I remember working with someone quite a while
ago (you?) And finding this problem but I never fixed it. I'll be sure to
get to the bottom of it this time.

DavidZemon, 2016-01-09:
Okay, the CMake package for OSX should contain the binaries, but I had the path incorrect in the documentation. TeamCity is rebuilding the documentation now, but here's the important snippet that I updated:

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~{.sh}
wget http://david.zemon.name:8111/repository/download/PropWare_Release20/.lastSuccessful/PropWare-2.0.0-Generic-osx_cmake.zip?guest=1 -Ocmake.zip
wget http://david.zemon.name:8111/repository/download/PropWare_Release20/.lastSuccessful/PropWare-2.0.0-Generic-propware.zip?guest=1 -Opropware.zip
wget http://david.zemon.name:8111/repository/download/PropWare_Release20/.lastSuccessful/PropWare-2.0.0-Generic-examples.zip?guest=1 -Oexamples.zip
unzip cmake.zip
unzip propware.zip
unzip examples.zip
sudo mv OSXCMake /usr/local
sudo mv PropWare /usr/local
sudo ln -s /usr/local/OSXCMake/CMake.app/Contents/bin/cmake /usr/local/bin/cmake
sudo ln -s /usr/local/OSXCMake/CMake.app/Contents/bin/ccmake /usr/local/bin/ccmake
sudo ln -s /usr/local/OSXCMake/CMake.app/Contents/bin/cmake-gui /usr/local/bin/cmake-gui
sudo ln -s /usr/local/OSXCMake/CMake.app/Contents/bin/cpack /usr/local/bin/cpack
sudo ln -s /usr/local/OSXCMake/CMake.app/Contents/bin/ctest /usr/local/bin/ctest

\# OSX only
sudo ln -s /usr/local/OSXCMake/CMake.app/Contents/bin/cmakexbuild /usr/local/bin/cmakexbuild
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

Notice how `CMake.app/Contents` has been added to the target path during the link phase.

DavidZemon, 2016-01-09:
Problems found and should be fixed now. Give the documentation another read and re-download the CMake zip and tell me how it goes.

dgately, 2016-01-13:
Ran through download/install/test process from: http://david.zemon.name/PropWare/Download.xhtml

New cmake.zip is a complete app (cool), installs without issue!
I was able to run through several example builds, without issue!

Closing this issue...

Thanks!

closed 2016-01-13 by dgately

DavidZemon, 2016-01-13:
Thanks so much for updating these tickets, and I'm _thrilled_ to hear you have it working! Perhaps I'll add a simple shell script to run through the steps provided on my the downloads page as part of a 2.0.1 release