parallaxinc/PropWare #61: Getting CMake errors when attempting to build the GettingStarted example on recent Mac OS X

dgately, 2016-01-01:
Getting: "Detecting CXX compiler ABI info - failed, on Mac OS X

On attempting to execute the PropWare download/installation instructions on a recent Mac OS X system (10.11.2 - El Capitan, latest as of 01/01/2016), I've run into a number of issues that I'm working-through in order to assist in your creation of an install package for Mac OS X. Here's one of the errors I get as I start to attempt a first build.

Note: my config places the propellergcc binaries within SimpleIDE app...

Portion of the output:

-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/SimpleIDE.app/Contents/propeller-gcc/bin/propeller-elf-gcc
-- Check for working CXX compiler: /Applications/SimpleIDE.app/Contents/propeller-gcc/bin/propeller-elf-gcc -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - failed
-- Detecting CXX compile features
-- Detecting CXX compile features - failed
CMake Error at /Applications/CMake.app/Contents/share/cmake-3.3/Modules/FindPropWare.cmake:301 (string):
  string no output variable specified
Call Stack (most recent call first):
  /Applications/CMake.app/Contents/share/cmake-3.3/Modules/FindPropWare.cmake:409 (set_compile_flags)
  /Applications/CMake.app/Contents/share/cmake-3.3/Modules/FindPropWare.cmake:433 (set_propware_flags)
  CMakeLists.txt:6 (create_simple_executable)
-- Configuring incomplete, errors occurred!

Full logs:
[CmakeError.txt](https://github.com/DavidZemon/PropWare/files/76017/CmakeError.txt)
[CMakeError_log.txt](https://github.com/DavidZemon/PropWare/files/76018/CMakeError_log.txt)
[CMakeOutput.txt](https://github.com/DavidZemon/PropWare/files/76019/CMakeOutput.txt)

DavidZemon, 2016-01-01:
Since I'm not handling version numbers very well yet (proper versioning will start after the final release 2.0.0), can you paste FindPropWare.cmake into this issue?

According to the log it is located in the following path:
/Applications/CMake.app/Contents/share/cmake-3.3/Modules/FindPropWare.cmake

I'm not seeing anything in the latest release-2.0-nightly branch around line 301 that would cause an error like that.

dgately, 2016-01-02:
Here it is...

[FindPropWare.cmake.zip](https://github.com/DavidZemon/PropWare/files/76090/FindPropWare.cmake.zip)

DavidZemon, 2016-01-02:
Ah, I very recently fixed that bug - the commit was made on 12/20/2015 and then merged in with the release-2.0 branch on 12/22/2015. Try downloading the latest version and give it a shot. The other fix is to simply provide the `MODEL` variable explicitly. It is supposed to support using LMM by default, but that was broken up until the aforementioned bug fix.

dgately, 2016-01-13:
Ran through the updated instructions at: http://david.zemon.name/PropWare/Download.xhtml

I was able to download, install and run through several of the examples!

dgately, 2016-01-13:
Mac OS X downloads/instructions appear to work quite well now.

Thanks!

closed 2016-01-13 by dgately