parallaxinc/PropWare #96: build instructions

krnlyng, 2016-09-12:
Hi,

can you please provide build instructions? I have tried various branches of this project and i get loads of errors.

For example on the development branch it doesn't find the PropWareConfig.cmake file.
https://bpaste.net/show/7f4e95852ee8

On the release-2.0 the file actually seems to exist but i get different errors:
https://bpaste.net/show/fdeb1fc1c9a0

Thanks in advance.

DavidZemon, 2016-09-12:
Sure, thanks for opening this. In the meantime, both of those errors are because you need to pull the submodules. Sorry, I'm on my phone so I'm not too much help at the moment

krnlyng, 2016-09-12:
Ah sorry, i was obviously too fast. After checking out the submodules everything built fine. Thanks

DavidZemon, 2016-09-12:
Glad to hear it! I'll get a new page added to the docs that describe this step, as well as what packages are required on the system.

DavidZemon, 2016-10-21:
@krnlyng, the new page is located here: http://david.zemon.name/PropWare/#/reference/build-from-source

I'd love to hear your thoughts on this page if you have any.

krnlyng, 2016-10-21:
thank you very much looks it is nice and clean!

closed 2016-10-21 by DavidZemon