parallaxinc/PropWare #108: UARTRX Read-Timeout

seijikun, 2016-11-29:
Would a timeout for `UARTRX`'s read methods be practical / easy to implement?
I've unfortunately absolutely no idea of how serial communication works.

A quick look at the source revealed `shift_in_data` as the method that does most of the work. It seems to use one of the io-wait methods (`waitpne` / `waitpeq`) to await the start of a transmission, which would probably have to be changed for a timeout functionality.

Any ideas on this topic?
My use-case is a duplex-communication with only 1 cog left. :grin:

DavidZemon, 2016-11-29:
Definitely possible. Not even hard. But it would significantly decreased the max baud rate, which is why i avoid it. What baud rate do you need?

seijikun, 2016-11-29:
2048B/s would be enough ; every B/s more is a pleasant extra.

Maybe not changing the current implementation, but adding a second one supporting timeouts would be a good idea.

seijikun, 2016-11-30:
#110 is a first approach to this.

DavidZemon, 2017-02-19:
@seijikun, please let me know when you've switched your codebase back to PropWare's develop branch and confirmed it is still working well for you. Once you can confirm that, I will close this ticket.

seijikun, 2017-02-20:
I've run the code for a couple of hours now. Up till now, everything seems to work just fine.

closed 2017-02-20 by DavidZemon