parallaxinc/PropWare #112: Can we expect updates or is it dead?

d0narnbl1tz, 2017-02-04:
also why do i get from http://david.zemon.name/PropWare/#/download
"PropWare-3.0.0.117-Generic" when the latest release here is v2.1.2?

Greetings
Daniel

DavidZemon, 2017-02-04:
Thanks for checking in. PropWare certainly isn't dead, and there will be updates hopefully in the coming months. Back in June of 2016 I picked up a second, very large project for my day job and it has severely cut into my home life. Coincidentally, I present that project this coming week! So, following that, I expect to get back to work on PropWare.

The 3.x branch in still in development, but reasonably stable. Therefore, I recommend any new users use that branch. Because there is no "stable" 3.x release, it means that I may change the API at any time without regard for backwards compatibility. I am following [semantic versioning](http://semver.org/), so expect numerous breaking changes between the 3.x and 2.x branches. Most of those changes have already been completed in the 3.x branch, but there are a couple tickets still left open that I may or may not finish. Particularly, #83 isn't marked as complete yet. I've broken the UART object up, but haven't decided yet if I'm going to break up SPI. I tried once, but it made a real mess of the SD card and filesystem objects. I2C almost certainly will not be broken up. #68 is also still "open" but I don't think there will be any more changes on that one. I just haven't put enough thought into yet to know for sure.

d0narnbl1tz, 2017-02-06:
Thanks for answering that fast.
Glad to hear that it is still alive.

closed 2017-02-19 by DavidZemon