parallaxinc/PropWare #117: broken links

bradleypeabody, 2017-03-16:
Looks like an upgrade to david.zemon.name broke some documentation links.  E.g. https://raw.githubusercontent.com/parallaxinc/PropWare/develop/README.md has links like: http://david.zemon.name/PropWare/GettingStarted.xhtml which appears to have changed to: http://david.zemon.name/PropWare/#/getting-started

My suggestion would be to add redirects for at least the old Download and Getting Started URLs to 301 over to the new location.

DavidZemon, 2017-03-17:
Thank you. Good point about the redirect - I'll get that going at least for the links listed in the README.

DavidZemon, 2017-03-17:
I didn't bother redirecting every page individually... sorry. Sometimes I'm just lazy. But I did put a regex in there to redirect anything *.xhtml to the new homepage.

bradleypeabody, 2017-03-17:
Gotcha, makes sense.  My main concern was that when I found a few broken links it looked to me initially like the project might have been dead - even the links land on a page that isn't exactly what you wanted, as long as it's clear where the new documentation is, I'm sure it will be fine.  P.S. Thanks for fixing the download links (e.g. http://david.zemon.name:8111/repository/download/PPropWare_Develop/.lastSuccessful/PropWare-3.0.0.132-Generic-osx_cmake.zip?guest=1 -Ocmake.zip), I think I just had really bad timing and ran into the link issue above and then and also happened to also try downloading while the build system was having issues earlier today.

Everything looks great now though.

DavidZemon, 2017-03-17:
Yea, bad day indeed. My server was migrated this weekend and a mistake was made during that migration, which is what brought down TeamCity (and therefore the download link) until the problem was brought to my attention a couple hours ago.

Btw, I notice you're looking at the osx package. There has been very limited testing of PropWare on Mac, so do please feel free to submit any new tickets here for help getting it setup. I have a label here on GitHub already setup for "questions" so its fine to report things here that are not bugs or feature requests.

DavidZemon, 2017-03-18:
Well that didn't work out too well. I just broke the existing doxygen docs - they all redirect to the homepage now. Better come up with a different solution.

avsa242, 2017-12-23:
@DavidZemon 
Not sure if this is because of the same issue, but none of the download links at the page referenced in the readme work:
https://david.zemon.name/PropWare/#/download

DavidZemon, 2017-12-23:
Thanks for letting me know. The only thing I can figure out is that TeamCity must have changed the download URL for artifacts with the latest release. I'm getting it fixed now.

DavidZemon, 2017-12-23:
Okay... working on getting this fixed. A few things all went wrong again. Most are fixed, but the links for the DEB and RPM packages are broken because they have been named incorrectly in the latest build. I'm working on getting that fixed. In the meantime, if you need the artifacts, please just download them directly from TeamCity: http://david.zemon.name:8111/viewLog.html?buildId=1791&buildTypeId=PropWare_Develop&tab=artifacts

avsa242, 2017-12-23:
> viewLog.html?buildId=1791&buildTypeId=PropWare_Develop&tab=artifacts

Awesome - quick turnaround, thanks!

DavidZemon, 2017-12-29:
Okay, all download links now fixed.

commit cb34ea90 (2017-03-17): [Closes #117] Fix broken links
closed 2017-12-29 by DavidZemon