parallaxinc/PropWare #118: Improve getting started documentation (specifically for OSX)

bradleypeabody, 2017-03-17:
I tried getting up and running with PropWare on OSX (following http://david.zemon.name/PropWare/#/getting-started ) and was able to get things working; but there are two important points which I basically had to just guess as I didn't see them covered.

The issue is that the build setup does not appear to know where to look for either the PropWare libraries, or the propeller-gcc binaries (and for the latter the error message is somewhat misleading because instead of finding propeller-gcc it finds clang and then complains about how some of the options being passed to the compile command are wrong... because it's the wrong compiler).

Setting these two environment variables *before* running 'cmake -G "Unix Makefiles" ..' solved it:

    export PROPWARE_PATH=~/Downloads/PropWare/
    export GCC_PATH=/Applications/SimpleIDE.app/Contents//propeller-gcc/bin/

It may be that I just don't know where to put these two items on my system and possibly including that information would help also.  But I think either way making it clear that these are the environment variables that control where to find these items, with an example, would be very helpful for the next person who comes along and tries to do what I just did.

DavidZemon, 2017-03-17:
I definitely see what you mean. Not only are those two variables missing from the Getting Started instructions, they aren't even mentioned in the [CMake Reference](http://david.zemon.name/PropWare/#/reference/cmake-reference) page. I'll get this updated.

Also, now that I _know_ the default installation for PropGCC on Mac, I can add that as a default search path, so you won't need to specify it in the future. So long as PropWare is installed somewhere other than what's provided on the website though, you'll always need to provide that environment variable.

bradleypeabody, 2017-03-17:
Okay great and that all makes sense.

One other suggestion on this would be to explicitly recommend a location for the PropWare libraries on the [download page](http://david.zemon.name/PropWare/#/download).  From what can I tell this isn't necessary on the other platforms because the installers do all the work.  But having a line in there or two that simply says something like "by default the PropWare libraries are installed into ... on Unix-like systems and ... on Windows; if you are not using an installer then either put PropWare into the appropriate directory for your system or set the PROPWARE_PATH environment variable to point a custom location".

(And similarly for propeller-gcc I would also suggest giving both the default location(s) that are searched for as well as the name of the env var to change.)

Something like that I think would make it very clear what to do, without being overly complicated either.