parallaxinc/PropWare #121: Best way to hack on PropWare for debugging things?

bradleypeabody, 2017-03-29:
I've run into a situation a few times now where I'm using the normal (pre-built) download of PropWare and using the CMake build setup as describe in the Getting Started guide and would like to make minor changes in the PropWare library and build my project with them as part of testing and debugging.

Example: I'm currently debugging some transmission issues with the mcp2515 CAN driver - it enqueues messages for transmission but doesn't actually send them (the send fails and the error I get is that it timed out, but more information is available in one of the registers).  In order to see what is happening I need more output from the guts of some private functions inside mcp2515.h.  I don't see an easy way to do that. Modifying the .h files appears to make no difference, presumably since it's getting all of the function definitions from the .a files.

Is there a simple solution to this?  Something I can toggle in a build file and have it take into account modifications to .h files in PropWare?

DavidZemon, 2017-03-29:
Short answer: no.

Long answer: it should work exactly the way you're doing it for `MCP2515` because it is a header-only object. It should not exist in the archives (the .a files). For those objects that _are_ compiled into the archives, I'll need to do some thinking about whether the best course of action is to recommend a process change on the part of the user, or a feature on the part of PropWare.

bradleypeabody, 2017-03-29:
Interesting, alright thanks for the quick answer.  Looks like I had a dumb issue with CMake cache and it was pulling from a different dir than that I thought and I was editing the wrong file.

You're right, it works fine to just edit the .h files for most things.

DavidZemon, 2017-03-29:
Btw, that MCP2515 implementation is very incomplete. I started working on it for a project that I thought I would do, but then dropped the project shortly thereafter. If you make some good changes, I'd love to review a pull request from you.

DavidZemon, 2017-08-22:
I'm going to go ahead and close this. Feel free to re-open if you have further questions.

closed 2017-08-22 by DavidZemon