parallaxinc/PropWare #123: question about file reading

bradleypeabody, 2017-04-25:
I'm trying to read data from an SD card as quickly as possible in sensibly-sized chunks.  Speed is very important in this case because it's for audio data and playback will suffer if things can't keep up.

I have this basic read loop that functions correctly but is (relatively speaking) very slow, presumably due to the amount of code that executes between retrieval of each character:

    SPI sdSpi;
    const SD driver(sdSpi,
        Port::Mask::P11, // mosi
        Port::Mask::P10, // miso
        Port::Mask::P9, // sclk
        Port::Mask::P12 // cs
        );
    FatFS filesystem(driver);

    ErrorCode err = 0;

    err = filesystem.mount();
    if (err) {
        pwOut << "Error mounting file system: " << err << "\n";
        return 0;
    }

    FatFileReader reader(filesystem, "sample.wav");
    err = reader.open();
    if (err) {
        pwOut << "Error opening wav file: " << err << "\n";
        return 0;
    }

    while (!reader.eof()) {

        char v;
        err = reader.safe_get_char(v);
        if (err == File::EOF_ERROR) {
            break;
        }
        if (err) {
            pwOut << "Error while reading wav file (" << c << " bytes read so far): " << err << "\n";
            return 0;
        }

        // not doing anything with the data just yet...

    }

I'm looking for a read path that allows me to say "read the next 512 bytes to this buffer".  The exact requirements would be:

* Read into a specific location in memory - in this case I need to be able to have two buffers and the audio playback will read from one while the SD read will target the other and then they swap.
* Doing this without any unnecessary copy would be important.
* I need to know when I've hit EOF - the code above does tell me that but even if I get direct access to the internal buffer being used by FileReader for example, I didn't see how to tell where the file ends.

Any ideas on how best to achieve that?  I've looked through the code a fair amount and it looks like FatFs has a buffer internally which ends up getting used as the target for the reads but it's private and seems to suffer from the above issues.

DavidZemon, 2017-04-25:
Step 1 is to instantiate a dedicated buffer for the file reader. Using the default will share with the fat table and cause unnecessary reloads.

(Typing on my phone, so I'll make a bunch of small comments)

DavidZemon, 2017-04-25:
Actually... That makes this whole thing a ton easier. See [this example](https://david.zemon.name/PropWare/api-develop/FileWriter_Demo_8cpp-example.xhtml) for using a dedicated buffer. Then you have access to  512 raw bytes. Once those 512 bytes are transmitted, you can read in the next block with [seek(512, seekdir::cur)](https://david.zemon.name/PropWare/api-develop/classPropWare_1_1File.xhtml#a7a2f36c9d4f8f85b2563c23a7c2573f6). You'll need to keep track for yourself whether or not you've reached the end of the file by getting the total length with [get_length()](https://david.zemon.name/PropWare/api-develop/classPropWare_1_1File.xhtml#a222ce3759375ae3ca34e9250c6dedbc4). This doesn't help you use a rotating buffer though... Which is a pretty significant oversight on my part if that's really unsupported. I thought there was a way to do it but I'm not finding it

DavidZemon, 2017-04-25:
I take back what I said about the default sharing with the fat table buffer. The fat table always gets its own buffer. But creating a dedicated buffer still gives you access to the raw bytes

DavidZemon, 2017-04-25:
```
Char array1[512], array2[512];

Buffer.buf = array1;

While (...) {
  WavPlayer.write(512, array1); // from cog 2
  Buffer.buf = array2;
  F.seek(512, seekdir::cur);
  Shared flag = 2;
  WavPlayer.write(512, array2); // from cog 2
  Buffer.buf = array2;
  ...
}
```

Psuedo code on a phone... Not easy :) but what do you think of this?

bradleypeabody, 2017-04-25:
Thanks for the replies.  Yeah, that does make sense.  The other approach I was thinking was to have two instances of FatFileReader, each with their own buffer and alternately seek and read from each one.  But if your idea above of swapping the buffer out between reads works, I think that's a simpler and better approach.

I will give this another shot soon and let you know how it goes.

bradleypeabody, 2017-04-25:
I'm still tinkering with this, but it looks like this is going to work out just fine.  The only PropWare change needed so far was to make FatFs.get_buffer() public, which allows one to do things like:

	BlockStorage::Buffer *pbuf = filesystem.get_buffer();
	pbuf->buf = sndBuf1;
	// do some reading, then sndBuf1 has the raw data I need
	pbuf->buf = sndBuf2;
	// do some more reading into sndBuf2, etc.

Keeping track of the read position manually is not a big deal and a workable approach.

I'll try to get everything fully working and then give you PR, assuming you're cool with a change that involves making get_buffer() public.

DavidZemon, 2017-04-25:
I might make this change myself tonight. I'll need to look through the `Filesystem` code and relearn a few things before I do, but I'll let you know what I come up with.

DavidZemon, 2017-04-26:
Having now had a chance to look through the code, I would discourage you from changing the default buffer in the `FatFS` class. I suppose it may not have any practical affect on your app, but I don't like that, upon invoking `FatFS::unmount()`, it will deallocate that shared buffer. To keep proper memory management, you'd have to ensure that the correct block of memory is set on the `Buffer` before invoking `unmount()`. Instead, I think it would be better to just provide a dedicated buffer to the `FatFileReader` constructor which can be statically allocated.

That being said, it's not the end of the world to change it to public. It would save you 512 bytes of precious memory, and the `FatFS` class only needs that buffer to search directory entries when finding a file, so it would be unused after that anyway. And by the time you get to memory deallocation, you probably don't care about the system state anyway.

So here's how I see it, three options:

1. You like my suggestion and just inject a dedicated buffer into `FatFileReader`'s constructor
2. I take your suggestion and make `Filesystem::get_buffer()` public (actually, I'd probably add a preprocessor macro like `PW_PUBLIC_FS_BUFFER` to switch that method between public and protected, that way users don't accidentally start using it without reading some important documentation about the memory allocation)
3. I add an overloaded `FatFS::mount(...)` method with two extra parameters. The signature for which might look like this: `PropWare::ErrorCode mount (const uint8_t partition = 0, uint8_t *directoryTraversalBuffer = NULL, uint8_t *fatBuffer = NULL)`. This would allow you to insert _your own_ statically (or dynamically) allocated buffers into the `FatFS` class and you would no longer need the `Filesystem::get_buffer()` method to be public.

What do you think?

bradleypeabody, 2017-04-26:
Option 1 is generally a good idea, but if there's one thing that is in scare supply in this environment it's memory; the idea of being able to have exact control over what buffers are allocated sounds very desirable.

On option 3, that sounds pretty good, but don't we need to pass in BlockStorage::Buffer* in order to allow the caller to swap out the pointer to the raw bytes?  This would require changing FileSystem::m_buf to a pointer instead of the actual struct value, but I don't see another way with this approach to allow swapping the underlying buffer.  Am I missing something?

DavidZemon, 2017-04-27:
Okay, it took me a long time to figure our my thought process, but here you go:

```cpp
const SD driver;
uint8_t buffer1[driver.get_sector_size()];
uint8_t buffer2[driver.get_sector_size()];
FatFS    filesystem(driver);
filesystem.mount(buffer1, NULL, 0);

BlockStorage::MetaData fileBufferMetadata;
BlockStorage::Buffer fileBuffer = {
        buf: buffer1,
        meta: &fileBufferMetadata
};

FatFileReader reader(filesystem, "fat_test.txt", &fileBuffer);
reader.open();
```

Here's the changes I had to make to `FatFS` to make this work: https://github.com/parallaxinc/PropWare/commit/a30454cb28c5d8e095c9d31241047cdaa9915e78

DavidZemon, 2017-04-27:
I'm not sure this is any better than option 2.... perhaps the only difference is that it makes the author do more work, therefore making it more obvious to the author that something funny is going on. This is starting to seem pretty silly and maybe option 2 really is better. Thoughts?

bradleypeabody, 2017-04-27:
I guess the question is what could go wrong by letting the author have access to the internal buffer.  Certainly things could explode if you change m_buf.buf and then let the thing get deallocated; however as you mentioned before as long as things are clearly documented it should be okay.

As a user of PropWare, it would strike me as expected behavior to read a comment on get_buffer() that says something like "You can use this to gain access to the internal buffer used for reads and write.  If you change the value of buf, subsequent reads or writes will use your new buffer - however you must take care to change it back afterward because the destructor will call free(buf)."  In a constrained environment like this, that all makes perfect sense - the guts are being exposed a bit to allow for performance optimization (avoiding a copy), a good and necessary tradeoff in this case.

That said I don't think you need to revert your changes - I see no problem with both approaches being possible (provide your own buffer or get access to the preallocated one).  But yes, I think get_buffer() being public and just having a good documentation comment is still a good idea.

DavidZemon, 2017-04-28:
Alright... another solution... I came up with this solution while I was still awake and thinking straight, but the implementation happened mostly after my brain turned off... so forgive me if there is a gaping whole somewhere. But the project does compile at least. What do you think of this? There is no more shared buffer in the `FatFS` class. It was only ever used during the mounting step anyway.

Not having a default buffer makes the simple case a bit more complex, but before I merge this back into develop I'll find some way to add a default back in.

DavidZemon, 2017-04-28:
Now, the real benefit to this new solution is yet unrealized: getting rid of the `malloc()` invocation in `mount()`. I now have only a single invocation left and I'm going to be working very hard to get rid of that tomorrow when I try to reimplement a default buffer. That would give everyone the best of every world

bradleypeabody, 2017-04-28:
Yes!  I totally agree with this approach - if the internal buffer can be removed entirely and replaced with something provided by the caller, and yet the simple case is still simple enough, that totally sounds like what we want.

DavidZemon, 2017-04-29:
It's been merged into develop and a new build is available. The result is exactly what we'd hoped. The FileReader_Demo size has decreased from 20,608 bytes (+ 1,024 bytes dynamically allocated = 21,632) to 21,308 bytes with no dynamic allocation. That's a 324 byte decrease in code size and an increase in functionality.

bradleypeabody, 2017-05-01:
This is awesome. I will have a look soon; thanks a lot.

DavidZemon, 2017-05-10:
How did the new solution work for you? Any further recommendations or requests?

bradleypeabody, 2017-05-11:
Thanks for the followup.  This appears to be exactly what I need.  Sadly my schedule has been pretty volatile recently and I haven't been able to get back to this in earnest yet and finish fleshing out all the related code, although I hope to soon.  But I think we're good on this, from everything I can tell.   Also, if I do end up writing classes that would be generally useful for audio playback and I can send you a PR with this so you can have a look and see if you want to include them.  I did get the beginnings of some PWM-duty-mode DAC code working and I also have an I2C DAC chip (https://www.sparkfun.com/products/12918) I'm trying out and between the two examples it would be good to find an interface that allows playback on either one and allows other implementations to be worked in as well.  One of the issues I've run into is that audio playback and the signaling involved approaches the maximum throughput of the Propeller.  44.1KHz audio, with an 80MHz processor corresponds to about 1800 instruction cycles per sample.  On the DAC chip, the I2C speed is prohibitive.  On the PWM output it's just not a lot of granularity (somewhere around 8-bit as far as I can tell) and the signal is less precise.  It means that my project may not work out on this platform :( , but I'm not sure yet.  It's possible I could do some crazy stuff like drive the audio output with multiple cogs, I'll have to see.

But yeah, on file system stuff, I think we're golden 👍

DavidZemon, 2017-05-29:
Are you generating the output on-chip, or reading from a pre-generated files (like wav/mp3/etc)? If you're just reading from a file, it might be best to do get a standalone audio chip that can decode the file. Funny store: that's actually what started PropWare: I had to build an MP3 player as part of a college course that used an SD card with mp3 files and an I2C MP3 decoder chip.

closed 2017-05-29 by DavidZemon

bradleypeabody, 2017-06-01:
Interesting story :) 

Good point, and I will look into that (yes, the application that started this issue involved playback from an SD card, not chip-generated audio).  One thing that makes it a bit interesting is that I also need the ability to record, which is probably going to mean uncompressed WAV files, not MP3.  I think it's going to be a bit more difficult to find a hardware solution for that sort of thing.  But I'll take a look.

DavidZemon, 2017-06-01:
Thought about a Pi Zero instead of or along side a Propeller? The Pi Zero would be cheaper than the MP3 decoder and could take a large load of the Prop. Not sure what your real-time requirements are, but maybe you don't even need the Prop at that point.

bradleypeabody, 2017-06-01:
Yeah, I know what you mean.  One thing the Prop really shines on in comparison to a Pi is power consumption, and it also has a pretty decent price point and physical footprint.  This particular application is supposed to be battery powered and last for a long time on one charge/set of batteries, which is basically a total bust on a Pi. Pis are in the hundreds of mA just when powered on and there is no sleep mode, and bootup takes ~15s at best.  That's the main thing for this application that made the Prop the right choice.  The project is stalled for lack of funding at the moment, but it might get picked back up, we'll see.   There is a separate audio project which involves a fair amount of DSP and doesn't have the same power consumption limitations - for that project I pretty quickly determined that Pi was a better fit.  Prop definitely has it's uses, each project has it's own requirements, just a matter of figuring out if the octagonal peg fits better in the round or square hole 😉