parallaxinc/PropWare #146: multiple definition of `_drive_speed'

BloodWorkXGaming, 2018-12-19:
When building PropWare with the include `#include <abdrive.h>` or `#include <abdrive360.h>` you get the following error from CMake:

```
C:\PropWare\share\PropWare\lib\cmm\libSimple.a(speed.c.obj): In function `_drive_speed':
(.text.drive_speed+0x0): multiple definition of `_drive_speed'
C:\PropWare\share\PropWare\lib\cmm\libSimple.a(drive_speed.c.obj):(.text.drive_speed+0x0): first defined here
collect2: ld returned 1 exit status
make.exe[3]: *** [SoPro] Fehler 1
make.exe[2]: *** [CMakeFiles/SoPro.dir/all] Fehler 2
make.exe[1]: *** [CMakeFiles/SoPro.dir/rule] Fehler 2
make.exe: *** [SoPro] Fehler 2
```

I suspect this is due to abdrive and abdrive both having the same functions (drive_goto, drive_speed, etc)

Any idea on how to fix that?

DavidZemon, 2018-12-19:
What version of the simple libraries are you

BloodWorkXGaming, 2018-12-19:
Ye, using the files being pulled in my PropWare.

I'm on Windows, Propware version 3.0.0.182, installed with the installer.

As a workaround I copy pasted the c files from the abdrive folder, and it is working just fine that way.

DavidZemon, 2018-12-19:
I'm able to build the attached project with the following size:

```
Code size  = 19412
Total size = 25200
```

If you run into the same error with the attached project, can you build with `VERBOSE=1`, such as `make VERBOSE=1` and then share the build output?

[abdrive_test.zip](https://github.com/parallaxinc/PropWare/files/2695340/abdrive_test.zip)

You are not able to reproduce the problem with the attached project, then please attach your project and I'll see if I can reproduce it with that.

DavidZemon, 2018-12-19:
Well that sure is interesting isn't it? That one function was used all over the place in the demo program and didn't have any issues.
I have it reproducing with your attached program... I'll dig deeper now.

BloodWorkXGaming, 2018-12-19:
Ye that surely is wierd...
Thanks for looking into it!

DavidZemon, 2018-12-19:
Long story short: there are obviously multiple definitions of the function in the Simple library. I'm seeing drive_speed.c in libarlo and libabdrive360 as well as libabdrive's speed.c. The correct fix is to get Parallax to rename all three functions to be unique (they do each have different implementations) but that could take anywhere from a day to a few years, depending on whether or not they care to support this use case.

So... I'm trying to find a work around. But now I have to go to work and make the real money :P

BloodWorkXGaming, 2018-12-19:
Thanks, that would make sense.

My current workaround is to copy the whole abdrive folder (with headers and c files) or abdrive360 respectively, whatever is needed. This works perfectly in my usecase.