parallaxinc/PropWare #154: Servo.h clashes with servo.h

BloodWorkXGaming, 2019-02-26:
The simpletools library `servo.h` can't really be used as there is a `Servo.h` library as well.
It is very mixed when which one gets included as the only difference is the capital S. (Works on one PC, doesn't on the other)

DavidZemon, 2019-02-26:
Oohhhh.... Oh Microsoft. That's annoying, for sure. Okay, I'll add a CMake option (which will default to `OFF`) to include Arduino's `Servo.h` (I'm guessing that's where the other one is coming from)

BloodWorkXGaming, 2019-02-26:
Awesome, thanks!

DavidZemon, 2019-02-28:
You're not trying to _use_ both Simple and libPropelleruino, are you?

DavidZemon, 2019-02-28:
@BloodWorkXGaming, please try downloading the version of PropWare available in either of the two builds linked above (click the green checkmark to get a link to the build on the CI server). This will let you test the change prior to me merging it in.

BloodWorkXGaming, 2019-02-28:
Well, yes, I am using both simple and libpropeller. Is there anything wrong with that?
serial.h from libpropellor (as that is waaaay better than fdserial from simple) and abdrive360 + servo.h from simple as there is no equivalent in libprobellor for that.

Will the new commits and come back to you, thanks!

BloodWorkXGaming, 2019-02-28:
Okay, so just tested the build but it sadly doesn't fix the issue.
As all subdirectories are roots for the import it doesn't really change much as the compiler still can't decide.
If I could do something like `#include "Simple/servo.h"` that might help.

Also, due to those changes there is now a `PropWare/PropWare` and a `Libpropeller/libpropeller` folder

DavidZemon, 2019-02-28:
I forgot a pretty critical part: replace `find_package(PropWare REQUIRED)` with

```
find_package(PropWare REQUIRED
    COMPONENTS Simple LibPropeller PropWare ArdionoJson)
```

This will exclude LibPropelleruino (note: not libPropeller) which has `Servo.h`. The extra level of directories is necessary for CMake to have this kind of control. And though you and I would be fine with `<simple/simpletools.h>`, it would completely break compatibility with code written in SimpleIDE and the Simple itself (look at the include lines in `simpletools.h` and you'll see what I mean). 

Also, keep this in mind if you ever write your own library... Make sure the headers are in a subdirectory with some kind of distinguishing name.

BloodWorkXGaming, 2019-02-28:
Okay with the new line in there I get the error
```CMake Error at C:/PropWare/share/cmake-3.13/Modules/FindPropWare.cmake:146 (message):
  PropWare's LibPropeller component not available due to missing
  PropWare::LibPropeller_cog
Call Stack (most recent call first):
  CMakeLists.txt:2 (find_package)
```

DavidZemon, 2019-02-28:
Ah, that's I get for trying to think at 6am. Try

```
find_package(PropWare REQUIRED
    COMPONENTS Simple Libpropeller PropWare ArdionoJson)
```

And of course, the `PropWare` and `ArduinoJson` components are optional since you're not using them.

BloodWorkXGaming, 2019-02-28:
Just noticed I am indeed using libpropellerino for the `WString.h`, so simply excliding it didn't fix it.
Copied the serial.h and serial.c code I needed into my project as a workaround.

DavidZemon, 2019-02-28:
Oh... bummer. Alright, I'll come up with something better (I'm going to leave the componentized folders though, and that's another thing that really should have been done a while ago).