parallaxinc/PropWare #162: Installer replacing Cmake on linux

l0wl3vel, 2019-10-13:
Since Propware seems to require some sort of patched CMake, is it possible to just use the pwcmake program in /usr/pwcmake/ and not replace the distros CMake with the outdated and patched one that comes with Propware? It seems kind of unnecessary to do this and is pretty hacky.

DavidZemon, 2019-10-13:
What version of PropWare are you using? This _should_ have been fixed with #150, which was merged into the `develop` branch on Jan 20, 2019 (build 193 of v3.0.0).

l0wl3vel, 2019-10-13:
Version 2.1.0 from the github releases section.

DavidZemon, 2019-10-13:
Ah, yea I haven't made an "official" release of 3.0.0 yet, which is why you won't see it there, but it's definitely the version I'd recommend using. PropWare's website links to that version for downloading as well: https://david.zemon.name/PropWare/#/download

closed 2020-12-06 by DavidZemon